LightWare-Optoelectronics/SampleLibrary #3: How to Change Parameters

AsternA, 2020-10-18:
Hey Guys, 

These examples are great but it missing how to change parameters,

for example, how do i send the character '?" to the LiDAR to see firmware, or how to send ?L and so forth.

Please help me as i am unable to send requests and change address or measuring frequency

thanks

Rob2048, 2020-10-18:
Hi Astern,

May I ask which product and which communication interface you are using?

AsternA, 2020-10-18:
Hey Rob,

It's LW20/I2C

S/N: S20-18540

Thanks
Almog

Rob2048, 2020-10-19:
You can find an example here: https://github.com/LightWare-Optoelectronics/SampleLibrary/blob/master/lw20_arduino_i2c/lw20_arduino_i2c.ino

This sends just the '?' command, but can be extrapolated to any command in the LW20 manual.

closed 2020-10-19 by Rob2048