LightWare-Optoelectronics/SampleLibrary #4: Getting 20.6m all the way on sf30/d via usb serial

danaki, 2020-10-20:
Hello, I'm trying to run Pythong samples for reading distance thru binary protocol via Serial. It outputs 20.6 meters no matter where I point the device. At the same time older protocol publishing distance in text in similar fashion does respond with correct distance reading and LightwareStudio also works as expected, so device is not damaged. What could be the problem?

Rob2048, 2020-10-20:
Hi Yuri,

Please let me know which sample you used so I can check it.

danaki, 2020-10-20:
* SampleLibrary/binary protocol/sf30_lwnx/sf30_lwnx.py
* SampleLibrary/binary protocol/python_lwnx_serial/lwnx.py

danaki, 2020-10-21:
@Rob2048 can you please confirm?

Rob2048, 2020-10-21:
The command set for the SF30/D is slightly different to the previous SF30 devices. I have added a sample specifically for the SF30/D here: https://github.com/LightWare-Optoelectronics/SampleLibrary/blob/master/binary%20protocol/sf30d_lwnx/sf30d_lwnx.py

You can find the SF30/D command set here: https://support.lightware.co.za/sf30d

danaki, 2020-10-21:
Works like charm

closed 2020-10-21 by danaki

danaki, 2020-10-21:
Btw is that ok?
```
# Set update rate to 78 readings per second.
print('Set update rate')
executeCommand(port, 76, 1, [8, 0])
```

76 != 78

Rob2048, 2020-10-21:
Yes it is correct. If you look at the documentation the opcode for setting update rate is 76. The 78 readings/sec is value 8.

http://support.lightware.co.za/sf30d/#/command_detail?id=_76-update-rate

danaki, 2020-10-21:
Yes, I've just checked it out. Thought you have a typo in the comment. Thanks.