LightWare-Optoelectronics/SampleLibrary #6: Streaming on sf30d

danaki, 2020-10-27:
I'm experimenting with streaming mode on SF30D. This is what works:
```
self._execute_command(self.port, 29, 1, [255, 255, 255, 255]) # seems has no effect
self._execute_command(self.port, 30, 1, [1]) # starts streaming
```

Then, by repeatdly reading the port I'm getting responses like so:
```
[170, 0, 1, 40, 1, 25, 11, 45, 17]
```

Which appears a non-documented "40" command and 5th and 6th bytes are actual distance. I'm calculating it like so:
```
        firstFiltered = response[5] << 0
        firstFiltered |= response[6] << 8
        firstFiltered /= 100.0
```

Which gives 28.41m in this case.

The question is -- what are the 7th and 8th bytes? It doesn't look like strength because the number there is like 4397, even divided by 100 doesn't look strength. What is this number? And of course, what is the command "40" and how to correctly make streaming work (please add an example)?

danaki, 2020-10-27:
Sorry, forgot to say, I'm doing it on serial connection and also that trying to set update rate via
```
self._execute_command(self.port, 76, 1, [value, 0])
```

seems has no effect. I've tried different numbers from 5 to 9, the speed is relatively constant and very high. I guess this setting doesnt effect the speed in streaming mode. If so then what setting does?

I0x0I, 2021-01-25:
The last two bytes are checksum [[1](https://support.lightware.co.za/sf30d/#/serial?id=packets),[2](https://github.com/LightWare-Optoelectronics/SampleLibrary/blob/master/binary%20protocol/sf30d_lwnx_c/src/lwNx.cpp#L132)].

Command 40 seems like a variable length version of command 44, it's activated by passing 11 to command 30.

The following sequence works for me
```
//Set the output rate to 200010 readings per second. (Command 76: Update rate)
lwnxCmdWriteUInt8(serial, 76, 0)
// Set distance output to distance only: (Command 29: Distance output)
lwnxCmdWriteUInt32(serial, 29, 0x01)
// Enable streaming of point data. (Command 30: Stream)
lwnxCmdWriteUInt32(serial, 30, 11)
// Read distance
lwnxRecvPacket(serial, 40, &response, 1000)
```
Hope this helps.
Cheers.

Rob2048, 2021-01-26:
This command will choose what outputs are present when the device streams data:
(https://support.lightware.co.za/sf30d/#/command_detail?id=_29-distance-output)

```
self._execute_command(self.port, 29, 1, [255, 255, 255, 255]) # seems has no effect
```

Streaming can be started by writing 5 to the 30 command:
(https://support.lightware.co.za/sf30d/#/command_detail?id=_30-stream)

```
self._execute_command(self.port, 30, 1, [5, 0, 0, 0]) # starts streaming
```

This will start a stream of command 44 packets.
(https://support.lightware.co.za/sf30d/#/command_detail?id=_44-distance-data-in-cm)

There are a few non documented streaming commands, like the 1 and 11 as mentioned above. These are for production purposes.

closed 2021-01-26 by Rob2048