LightWare-Optoelectronics/SampleLibrary #16: SF45B Scan Speed vs Angular Resolution vs Update Rate

Mahmood-Hussain, 2025-04-21:
I have a question regarding the interpretation of the *Scan Speed* parameter in our system. I understand that it can take values ranging from 5 to 2000, but I am unsure of the unit associated with it. Specifically, I would like to understand what a value of `5` represents. 

To be more precise, I would like to determine:

1. **The time it takes to complete a single scan** when `Scan Speed = 5`.
2. **The number of points generated in a single scan**, given the following configuration:

```
[85] Scan Speed     = 5
[66] Update Rate    = 5000
[98] Scan Low Angle = -160 degrees
[99] Scan High Angle = +160 degrees
```

My goal is to allocate an appropriately sized array to store all points from a single scan. While allocating an array based on the update rate would be safe, it may lead to inefficient memory usage. Therefore, I would greatly appreciate any clarification on how the actual number of points per scan can be calculated from the above parameters.

Rob2048, 2025-04-23:
Hi @Mahmood-Hussain,

There is not a specific unit associated with the 'Rate' (scan speed) parameter. I've had a look and there is a fairly complex relationship between the scan speed and the behavior of the motor position.

However it will remain consistent when operating under normal conditions and with fixed parameters. You seem to have very specific angle/speed/update parameters in mind, so I have run some tests to get you some numbers.

With the parameters you have specified I am seeing ~2230 points per each unidirectional sweep. You'd be safe allocating for around 2300 points.

Take note that during special cases (like startup, or interruption to the motion) there will be a longer acceleration cycle and you will get potentially more points in a sweep.

Mahmood-Hussain, 2025-04-23:
@Rob2048 thanks I will maybe run experiments to try to get these values then.